klonyyy/MCUViewer #134: Core triggers hardfault handler

Diveturtle93, 2025-08-28:
### Version/Branch:

release 1.2.3

### Operating system:

WIndows 10

### Debug probe:

ST-Link2

### Microcontroller:

STM32F767ZIT

### Details:

Hello,

my controller triggers hardfault handler in different time intervals, if I use the logging via MCUViewer. I inserted three variable to the plot. The programm works fine for hours, if I didn't log it.

The minimum time interval, were only a few seconds. Maximum were a few minutes.

### Screenshots/Video:

[image]

klonyyy, 2025-08-28:
Hello, are these regular variables placed in RAM? What are their addresses? Sometimes hardfaults might occur if you're reading peripheral registers with MCUViewer.

klonyyy, 2025-09-19:
Closing due to inactivity - please reopen if needed.